JdeRobot/base #245: FlyingKinect world wrong definition

fqez, 2015-11-27:
Hi, 

the flyingKinect.world is wrong defined as it includes models outside the world tag, so instead this:

```
<?xml version="1.0" ?>
<sdf version="1.4">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://flyingKinect</uri>
    </include>
  <world name="default">
    <include>
      <uri>model://sun</uri>
    </include>
  </world>
</sdf>
```

we need this:

```
<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://flyingKinect</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
  </world>
</sdf>
```

varhub, 2015-11-29:
It raises a warning like "include tag was not expected, skipping it", or similar?

fqez, 2015-11-29:
No, it does not say anythig. It's also true that I didn't launch the world with the 'verbose' flag of Gazebo. I will tell you if something like that appear in the log files soon.

fqez, 2015-12-15:
Fix included in #270

closed 2016-01-19 by fqez