JdeRobot/base #260: Improvement on replayer and replayController

chanfr, 2015-12-08:
There are a couple of things that we need to improve in replayer and replayController Components:
1. Include a replayer status to know in a remove component the status of the replayer. The available satus will be:
   a. WAITING TO START
   b. PLAYING
   c. PAUSED
   d. FINISHED

2. Include a configuration key to the initial status of replayer basically:
   a. Loop replay active
   b. Start to play when the component starts

chanfr, 2015-12-08:
The following enum has been included into the recorder interface:
	enum ReplayerStatus { WAITING, PLAYING,PAUSED,FINISHED};

And the replayController interface has a new method:
		ReplayerStatus getStatus();

Also the replayer component accepts two new configurations:
              Replayer.startPlaying=1
              Replayer.repeat=0

varhub, 2015-12-09:
Hi @chanfr, if changeset is not closed, could you add default values to control.h constructor? Therefore api will be bakward compatible (if this makes sense)

ps: initial request was to add namespace (jderobot::replayer::X), but as long as it is a core feature, I even reverted my own though

closed 2021-02-21 by jmplaza