JdeRobot/base #261: Simplify the use of jderobot interfaces on the client side

chanfr, 2015-12-08:
We have to write too much code (and duplicated) while we are using the client side of the jderobot interfaces. The idea is to include a new handler to reduce all the duplicate the code and simplify the use of all the interfaces.

chanfr, 2015-12-08:
Using the example a replayController  we have to write:

	try{
		Ice::ObjectPrx base = ic->propertyToProxy(prefix+"control.Proxy");
		if (0==base){
			throw "replayController: Could not create proxy with replayControl";
		}
		else {
			prx= jderobot::replayControlPrx::checkedCast(base);
			if (0==prx)
				throw "Invalid " + prefix + ".Proxy";
		}
	}catch (const Ice::Exception& ex) {
		std::cerr << ex << std::endl;
	}
	catch (const char* msg) {
		std::cerr << msg << std::endl;
		std::cout << "replayController: Not camera provided" << std::endl;
	}

only to use the proxy of the interface.

chanfr, 2015-12-08:
Now using the commonHandler template you will only need to write:

     std::string replayControllerName = prop->getPropertyWithDefault(prefix+"control.Proxy", "");
     CommonHandler<jderobot::replayControlPrx> replayController(ic,replayControllerName,false)
	 if (!replayController->getProxy())
		exit(-1);

You will have the jderobot::replayControlPrx prx; under replayController->getProxy().

aitormf, 2017-03-10:
JdeRobotComm (#654) solves this problem among others. See #683 and #696

closed 2017-04-10 by aitormf