JdeRobot/base #266: refactoring of flyingkinect Gazebo plugin

jmplaza, 2015-12-12:
According to the improvements in the quadrotor plugin (#229) by @varhub . The main goal is to solve the time lag between image data and pose3D data delivered by current plugin. This variable delay hinders the map building from simulated RGBD data (for instance http://jderobot.org/Jnbosgos)

Hopefully it will also solve #245.

varhub, 2015-12-12:
Hi,

Here I can only see two options.
1. Provide a single struct that holds both data. Therefore, single plugin
style will make sense (because all except image is realized at model level).
2. Remove any intermediate topic and Ice threads moving towards a push
based interfaces.

(1) is pretty bad because obey to redo camera provider.
(2) can be not enough and some like external trigger will be required

varhub, 2015-12-12:
Yes, 245 will be in same changeset, but with milestone merging to simplify
revision.
@fqez, your task is comming.

varhub, 2015-12-15:
Almost done here: #270

fqez, 2016-01-04:
Merged!

PR #287 (2016-01-25): Split flyingKinect2 from source to header+source files
Complete closed #266
closed 2016-01-04 by fqez