JdeRobot/base #267: kobuki driver is not working on the real robot

jmplaza, 2015-12-12:
It was in the previous release of JdeRobot, one year and a half ago. We should revisit it and refine its integration in the new compilation chain.

varhub, 2015-12-12:
Please, provide the last version/tag/commit when it was working and first commit/version when issue was detected.

fqez, 2016-03-02:
The integration is done, the only thing todo is to upload the code to the Third Party repository and release the first package for kobuki_driver.

fqez, 2016-03-09:
Modified CMakeLists.txt ( #372). Removed OpenCV_LIBRARIES conditional to kobuki_LIBRARIES. Now it compiles if it finds the kobuki libraries.
Changed also the dependency from kobuki to kobukilib. The last one is the name of the kobuki libraries package (tested in the testing apt repository and working, soon in the official repo).
The next jderobot package will depend on kobukilib package, so it will include also kobuki_driver binary, that will allow to work with the real kobuki.

Testing the environment now...

fqez, 2016-03-10:
**Update**

Tested and working on a clean installation Ubuntu 14.04 x64 on virtual machine. jderobot now installs kobukilib and the kobuki_driver binary.

TODO:
 [x] Update kobukilib source to JdeRobot-ThirdParty
 [ ] Upload kobukilib package to the official apt repository
 [ ] Generate new jderobot package (5.3.1-rc6) with the most recent changes for both x86 and x64 architectures