JdeRobot/base #278: Is followball executable being installed?

varhub, 2015-12-22:
Referring to https://github.com/RoboticsURJC/JdeRobot/blob/dc9cd7c3022a8d84197ec378a701c6616293c8f2/src/stable/components/followBall/CMakeLists.txt#L59

Is this executable being installed?
I guess it isn't due executable do not match exactly with folder name (case sensitive).

Can anyone revise this file? (at /usr/local/bin)

fqez, 2015-12-26:
Yep, I can confirm this issue. This executable is not being installed at all due to the name convention, as @varhub said.

jmplaza, 2015-12-26:
Yes, we have some obsolete components in the github repository yet (followball among them). Maybe they should be moved to a different directory (testing/examples?), this could be part of #234. Some cleaning is desirable. Let's do it (together with the other tasks in #234) after the launching of the next JdeRobot official release

varhub, 2016-01-09:
If they work move them to another directory just to hide it do not worth it.
* testing tree is now almost deprecated or broken. So no one could reactivate this component in proposed directory
* each new directory at now requires a lot of tweaking and hacking at cmake, so these changes must be planed to life forever

There is a simple option: if followball do not seems to be part of the active framework, just extract it (git split) to allow on demand git clone and compilation.

*Related to #234*

fqez, 2016-06-22:
That component is going to be deprecated soon, so we will remove it from the repo, so I can close this issue!

closed 2016-06-22 by fqez