JdeRobot/base #284: Need to improve the Intrrfaces section from the.manual

fqez, 2015-12-29:
The current status of that section only has a link to the ice file in github. Some subsections are okay, but many of them are empty. The idea is to complete the section adding some multimedia content that helps the user to understand the basis of the interfaces he needs to use.

varhub, 2015-12-29:
Hi @fqez, is jut my opinion but I think that populate git with a lot of issues that do not affect to code and can not be tracked by git is dirty and do not provide a valuable workflow.

For me there are two options:
  1. Do a better and complete messages that also includes specific links to mediawiki and even those that point to current version. Therefore it will be pretty easy to see what was changed.
  2. Do a single issue (#271) for all these stuff and define each new task (like #272 #283 #284) as subtask of initial issues. Therefore, each new task is a tittled reply that can be referenced as #271-D

(2) remove changeset fragmentation, so anyone can see what was improved or created after #271, but also makes dirty discussion thread. By the way, who is replying?
Finally, you can transfer discussion and notification to community and/or devs maillist by just copy-paste message to email and send it. Doing it from chrome will preserve markdown format and even will handle hrefs.

fqez, 2015-12-29:
Hi @varhub,

I appreciate your observation about my posts and you are right in some aspects. I agree with you about the message style, maybe it should be more "pretty", but that does not change the fact that each task must have an issue associated, or at least that was I said to do. As long as the manual changes are incremental and pretty big, creating just a single issue for that does not seem correct to me so your second option makes no sense. It's also true that I'm being tracked by JoseMaria each week, so to facilitate our job it's better to create an issue for each change I make due to we can isolate them, I think.
Furthermore, issues are meant to describe problems or improvements ToDo, that means, not done yet, so I can't describe the made changes before they are done. I can do that whenever I close an issue, not before. So I may change my posting style adding links and stuff, and I may close the issues with the description of the changes made, as long as they have no PR associated to explain those changes. 
It seems that I am not as professional as you are when it comes to document things, but I will try to improve, thanks! :)

varhub, 2015-12-29:
Don't worry @fqez, it was just my opinion.
By add the two links (to page and to version X) is enough.

By the way think in this:
If you create an issue, you close it and no one reply or review the manual
and push a +1, it even exist? (if a tree fall in the forest...)

For tracking purposes (out of git) you can always use #x-A, #x-B and so on.
Therefore issue message is motivation and each reply a task.
Eventually a issue tagged as announcement could notify tasks' end and
summarize them or just link to all changes. Be warned that you may do this
end issue to gather all created issues into a context. Futhermore, all
these issues should have same title prefix to ease identification.

Do you understand why I was avoiding option 1?

fqez, 2016-01-05:
Following this at #286

closed 2016-01-05 by fqez