JdeRobot/base #286: Manual improvements

fqez, 2016-01-05:
# Mediawiki's manual updates changeset

Here you have the work done and ToDo's for the mediawiki's [manual](http://jderobot.org/Manual-5)

**Tasks list**

1. [x] Improvement of the [main page](http://jderobot.org/Manual-5)
2. [x] Structural improvement of:
    - [x] [Drivers section](http://jderobot.org/index.php/Drivers)
    - [x] [Interfaces section](http://jderobot.org/index.php/Interfaces)
    - [x] [Tools section](http://jderobot.org/index.php/Tools)
    - [x] [Running JdeRobot section](http://jderobot.org/index.php/Running_JdeRobot)
3. [ ] Content improvement of:
    - [x] [Drivers section](http://jderobot.org/index.php/Drivers)
    - [ ] [Interfaces section](http://jderobot.org/index.php/Interfaces) ***WIP***
    - [x] [Tools section](http://jderobot.org/index.php/Tools) [*Minor changes pending*]
    - [x] [Running JdeRobot section](http://jderobot.org/index.php/Running_JdeRobot) [*Missing some examples*]
4. [x] Update [Downloads section](http://jderobot.org/Downloads)

varhub, 2016-01-09:
Hi @fqez!, this provides better context, isn't it?

By the way, recovering #284 discussion I noticed that you cannot define a link with the whole changeset (mediawiki limitation?). At least, append each task with begin date worth it because it will allow to manually get these changeset without be lied with author nor dates.

For example: Running JdeRobot section (2015-12-17)
therefore one can get history and just compare with previous version of this date http://jderobot.org/index.php?title=Running_JdeRobot&diff=13955&oldid=12723

I did a minor reading, but I saw that (Image)[http://jderobot.org/index.php/Interfaces#Image] should be generic and current depth info must be in a subsection (It should be at ImageTypes, colorspacesmm or whatever, but by now a DEPTH_888 or similar title that wraps these info seems ok).

jmplaza, 2016-05-16:
For improving the Interfaces section of the web documentation, let's us focus on this basic set for now: motors, laser, pose3D, CMDvel, ArDroneExtra, images/camera.

aitormf, 2016-05-31:
Interface section done.

closed 2017-10-11 by aitormf