JdeRobot/base #289: new kobukiviewer component

jmplaza, 2016-01-05:
introrob, introrob_py and introrob-qt will soon be moved to TeachingRobotics git (https://github.com/RoboticsURJC/JdeRobot-TeachingRobotics), as they are more oriented to teach robotics that to be generic middleware. A component for teleoperating Kobuki and Pioneer wheeled robots and viewing their onboard sensors is required. 

Following the name convention KobukiViewer may be a good name for it. It will be mainly based in current introrob-qt, but without the "student callback" that takes the control of the robot when the PLAY button is clicked on the GUI.

varhub, 2016-01-05:
It is also a chance to do generic fragments (qt widgets) so both components can reuse it.
Note that both have almost same sensors and behavior so isolated development will be a mistake.
Eventually, one can even extend the other. By the way is preferable to do several little interfaces to ensure better desktop's area usage (faced to context lost)

fqez, 2016-01-05:
Can you be more specific when you talk about qt widgets? What functionalities should they have? I guess you are suggesting a "modular" implementation where you can customize your own component, am I right?

varhub, 2016-01-05:
I'm talking about a "cut and place" approach. Therefore an interface is a
composition of several puzle pieces.
Here arise some questions and problems, like threading and spinOnce.
For example, a image viewer, with the ice proxy and the spin ratio should
be a widget. Also "ball control": gui plus setcmdvel

Corolary: each principal ice interface should have a widget, with its own
ice prefix and proxy (this is the front end).

But if we can switch to javascript and therefore to html, html templating
or iframes could be also feasible

fqez, 2016-02-03:
Done, we can take a deeper look in other chance. Merged at #305

closed 2016-02-03 by fqez