JdeRobot/base #297: plugin flyingKinect2: interdependence with plugin quadrotor2 and instability during executioin

jnbosgos, 2016-01-11:
First, I noticed that for the compilation of the flyingKinect2 plugin, must be compiled the quadrotor2 plugin, because it has a dependency with the latter.

On the other hand, I have done tests with the flyingKinect2 plugin, the component navigatorCamera, and other own component, mapper3Drgbd [1]. Although the system connects well, during the execution of the tests usually the plugin crashes and it hangs the system.

[1] http://svn.jderobot.org/users/jnbosgos/pfc-teleco/trunk/components/mapper3Drgbd/

varhub, 2016-01-11:
This is very very bad.
Could you provide error output or a log about it? (run gzserver with --verbose)
Also when it happens and how many time it is working (to run it further than it).

varhub, 2016-01-12:
About compile, you can use buildpresets.
build_purge and test_flyingKinect are your friends.

jnbosgos, 2016-01-12:
About compile, I've seen #270 but with all respect what I do not understand is why I should compile a plugin that I will not be used (quadrotor2) to compile the plugin flyingKinect2; then I don't appreciate the relation between the two plugins (a quadrotor and a kinect).

As for the instability, the logs of the three elements when the system crashes are:

#################### Plugin flyingKinect2 ####################
jnbosgos@ubuntu:~/JdeRobot/src/stable/components/gazeboserver/plugins/flyingKinect2$ gzserver --verbose flyingKinect_GrannyAnnie.world 
Gazebo multi-robot simulator, version 5.1.0
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.35
Constructor CameraIBase
Constructor CameraIBase
[flyingKinect] KinectPlugin::LoadSensors()
[flyingKinect] KinectPlugin::InitializeIce()
[flyingKinect] 	config: flyingKinect2.cfg
[Info] loaded Ice.Config file: flyingKinect2.cfg
[flyingKinect] Ice adapter listening at 
[flyingKinect] 	tcp -h localhost -p 9997
[flyingKinect] KinectPlugin::Init()
[flyingKinect] 	bootstrap depth camera
PushCameraI::onCameraSensorBoostrap()
PushCameraI::onCameraSensorBoostrap()
Violación de segmento (`core' generado)

#################### component navigatorCamera ####################
jnbosgos@ubuntu:~/JdeRobot/src/stable/components/navigatorCamera$ ./navigatorCamera --Ice.Config=navigatorCamera.cfg
[INFO] Negotiated format  for camera 
Loading glade.
Glade done.

TcpTransceiver.cpp:309: Ice::ConnectionLostException:
connection lost: Conexión reinicializada por la máquina remota
[WARNING] navigatorCamera.CameraRGB.error during request (connection error)
--------navigatorCamera.CameraRGB. adquisition timeout-
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
navigatorCamera.CameraRGB.error during request (connection error)
^C

#################### component mapper3Drgbd ####################
jnbosgos@ubuntu:~/JdeRobot/src/stable/components/mapper3Drgbd$ ./mapper3Drgbd --Ice.Config=mapper3Drgbd.cfg
[INFO] Negotiated format  for camera 
Loading glade.
Glade done.
RANSAC CUSTOM Algorithm.
NONE Algorithm.
RANSAC CUSTOM Algorithm.
Control -> (cycle_control - diff) = -21158 us.
Control -> (cycle_control - diff) = -21639 us.
Control -> (cycle_control - diff) = -31456 us.
Control -> (cycle_control - diff) = -19768 us.
Control -> (cycle_control - diff) = -31065 us.
Control -> (cycle_control - diff) = -22812 us.
Control -> (cycle_control - diff) = -30086 us.
Control -> (cycle_control - diff) = -31094 us.
NONE Algorithm.
RANSAC CUSTOM Algorithm.
Control -> (cycle_control - diff) = -139907 us.
Control -> (cycle_control - diff) = -124054 us.
Control -> (cycle_control - diff) = -126767 us.
Control -> (cycle_control - diff) = -129487 us.
Control -> (cycle_control - diff) = -132270 us.
Control -> (cycle_control - diff) = -128806 us.
Control -> (cycle_control - diff) = -128053 us.
Control -> (cycle_control - diff) = -132573 us.
NONE Algorithm.
Control -> (cycle_control - diff) = -138340 us.
RANSAC CUSTOM Algorithm.
terminate called after throwing an instance of 'Ice::ConnectionRefusedException'
[WARNING] mapper3Drgbd.CameraDepth.error during request (connection error)
  what():  Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Conexión rehusada
Abortado (`core' generado)

#################### END LOGS ####################

It tends to fail when I'm moving the sensor.
Sometimes failure in a minute and sometimes endures more time working.

varhub, 2016-01-13:
I think bug was found:
Image dimensions are bad taken.

Testing source at https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 @ debug/check-static-memory

@fqez, help! ;)

varhub, 2016-01-13:
Bug not characterized, but workaround is here:

https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 @ hotfix/fallback-to-per-sensor-update

Try it and let me know

fqez, 2016-01-13:
Are you compiling gazebo with all the plugins enabled? Try a clean build disabling the others:

    cmake .-Dbuild_pioneer=OFF -Dbuild_quadrotor=OFF -Dbuild_flyingKinect=OFF -Dbuild_turtlebot=OFF -Dbuild_quadrotor2=ON -Dbuild_flyingkinect2=ON

varhub, 2016-01-14:
@jnbosgos, try @fqez sentence first.
About my patch, you must do:
```
cd src/stable/components/gazeboserver/plugins/
rm -r flyingKinect2 # or rm -fr flyingKinect2
git clone https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 -b hotfix/fallback-to-per-sensor-update flyingKinect2
# and then cmake (qtcreator recommended) and build
```
This create a new isolated repo for flyingKinect, so it will be pretty easy to fetch new patches if required.

jnbosgos, 2016-01-17:
Thank to both of you.

Since the beginning of this problem I have only compiled the needed plugin (flyingKinect2) and their dependencies (quadrotor2).

The exit log of the execution for the patch at https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 @ debug/check-static-memory, is:

<pre>
<code>jnbosgos@ubuntu:~/JdeRobot/src/stable/components/gazeboserver/plugins/flyingKinect2$ gzserver flyingKinect_GrannyAnnie.world
Constructor CameraIBase
Constructor CameraIBase
[flyingKinect] KinectPlugin::LoadSensors()
[flyingKinect] KinectPlugin::InitializeIce()
[flyingKinect] 	config: flyingKinect2.cfg
[Info] loaded Ice.Config file: flyingKinect2.cfg
[flyingKinect] Ice adapter listening at 
[flyingKinect] 	tcp -h localhost -p 9997
[flyingKinect] KinectPlugin::Init()
gzserver: /home/jnbosgos/JdeRobot/src/stable/components/gazeboserver/plugins/flyingKinect2/src/kinect.cc:209: void kinect::KinectPlugin::_on_cam_update_depth_data(const float*, unsigned int, unsigned int, unsigned int, const string&): Assertion `((void*)img_depth_raw.data == (void*)_data) && "STATIC_MEMORY UNMET"' failed.
Abortado (`core' generado)</code>
</pre>

And with the source code of the other patch at https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 @ hotfix/fallback-to-per-sensor-update, the problem with the instability seems to be solved. After several tests, it has not crashed.

Thank you again.

jnbosgos, 2016-01-25:
@varhub, I forgot to comment you if you could upload your patch at https://github.com/jderobot-varribas/gazeboplugin-flyingKinect2 @ hotfix/fallback-to-per-sensor-update to the official repository.

Thanks.

varhub, 2016-02-05:
Hello @jnbosgos , if you really need it can spend some time on it.
As you can see [here](https://github.com/RoboticsURJC/JdeRobot/pull/269#issuecomment-180468820) this is not more my competence, so...

Pehaps @fqez can take this task and commit it with the **--author** arg, just to be polite with developers.
You can always use previous version of flyingKinect or do a new one by yourself, then you will understand what means quadrotor2 dependency and why is mandatory.

closed 2017-02-23 by fqez