JdeRobot/base #300: support for Orbecc Astra RGBD sensor

jmplaza, 2016-01-14:
Hopefully it will work with our openniserver as it is.

lr-morales, 2016-01-20:
Tested against openniServer and openni1Server as they are from two diferent hosts, without good results.

OpenNI library seems to fail to identify the Orbbec Astra sensor as a valid device from different usb ports and hosts (`lsusb` was checked in all cases, an reported new devices when sensor was plugged in), so this problem should be investigated first before determining if an openniServer modification is needed. Next step in the issue resolution will be checking this problem.

chanfr, 2016-01-21:
Hi,
I've tested openniserver with several devices. One thing I had to do in many cases is to download the driver and copy it into the drivers folder (I cant remember the exactly the folder). It seems that the system is recognizing the device but openni does not have a properly driver to interact with it.  

Maybe the driver is compiled with their sdk and we can integrate it:
https://github.com/nocnokneo/astra

lr-morales, 2016-02-03:
Thanks for the tip about the driver.

After trying the precompiled version of the driver against the packaged OpenNI2 of Ubuntu, I compiled the complete SDK (https://github.com/orbbec/astra) and OpenNI2 version (https://github.com/orbbec/OpenNI2) from Orbbec.

Using the driver created with the OpenNI2 version of Orbbec, I could finally access to the device via OpenNI2 (at least to the depth image).

Next steps are now:
 - Getting the driver to work with OpenNI2 package or some kind of solution for JdeRobot
 - Retrieve successfully both RGB and Depth sources from OpenNI2 test tools
 - Make it work with JdeRobot RGBD components

lr-morales, 2016-12-12:
After reviewing some aspects of this issue, knowing about some proyects working about this device on ROS and viewing the advances on the JdeRobot compatibility with ROS drivers, the action route now changes to:

- Make one of the ROS drivers found work in ROS Kinetic with rviz
- Plan to make RGBDviewer a bilingual component, in order to be able to connect it to the ROS driver
- Check the connection

closed 2021-02-21 by jmplaza