JdeRobot/base #301: Support for electromagnetic parking sensors

jmplaza, 2016-01-16:
There are parking electromagnetic sensors in the market that use an anthenna to detect distance to obstacles near the car (under 1m approx). As output they use a buzzer to signal one of three levels on the measured distance. Supporting these sensors opens the door to new applications

lr-morales, 2016-02-29:
# Progress report
## Check list
- [x] Locate and acquire sensor for testing
- [x] Preliminary system analysis and working test
- [x] Study the output signal
- [ ] System response characterization
- [ ] Devise good connection method to RPi2
- [ ] Receive signal from sensor
- [ ] JdeRobot driver encoding
- [ ] Driver testing

## Locate and acquire sensor for testing

Searching around the net, finally decided to use one good price sensor system composed by a central unit, a buzzer and a metallic adhesive antenna band.

The antenna of this sensor system is intended to be fixed inside the rear plastic bumper of the car; the central unit and the buzzer are fixed with an adhesive inside the car. The central unit is designed to be connected to 12V - ideally to the reverse light - and a valid ground.

[image]

## Preliminary system analysis and working test 

For a first analysis, soldered a connector to the power input lines to allow a commercial 12V adapter to be used instead a real car; the antenna was partially unrolled on a table leaving a distance of more than 1m to other obstacles in one side.

After powering up the system, the buzzer beeps 3 times to signal that is ready to operate. If there is a bad connection, something wrong with or touching the antenna or another problem, the buzzer beeps with a sequence of deep tones.

When working in good conditions this behaviour can be observed:

- Total insulators (dry wood, cardboard, plastic) don't affect the sensor as it relies in the electromagnetic field detected by the antenna.

- Objects moving away from the sensor are ignored.

- If an object moves excessively near to the sensor (*few centimetres*), buzzer beeps in similar way as an error. It needs to rearm itself before operating again (*rearm time and conditions to be studied*).

- If an object moves too fast towards to the sensor, buzzers beeps signalling the excessive speed. Then readjusts the detection zone (*need to determine exactly this readjustment*).

- If an objects moves in a paced way towards to the sensor, 3 areas are defined (*distances to be determined experimentally*):

  - **Far area**: buzzer sounds beeps of medium pitch spaced proportionally to the distance to the sensor.
  - **Near area**: buzzer emits a constant high beep.  
  - **Contact area**:buzzer emits a constant deep beep.  

## Study the output signal

In order to connect the sensor to JdeRobot, a method to interface the sensor against some kind of computer is needed. The main output of the system is the buzzer, so a good approach is trying to adapt the signal sent to this buzzer to a device with JdeRobot.

Analysing the signal sent to the buzzer in a quick way with an oscilloscope we get this kind of results:

[image]

[image]

[image]

It seems to be a low frequency PWM-like signal in the range of 1~3V.

## System response characterization

TODO

## Devise good connection method to RPi2

TODO (*Working*)

## Receive signal from sensor

TODO

## JdeRobot driver encoding

TODO

## Driver testing

TODO

lr-morales, 2016-03-31:
Updated the first comment of this issue (#301) with the progress achieved.

closed 2021-02-21 by jmplaza