JdeRobot/base #302: cmake process broken in Debian Sid

rocapal, 2016-01-18:
Cmake process is broken in Debian Sid. The ouput is the following. Notice that there are two different issues when "cmake ." command is executed.

```bash
CMake Warning at Deps/opencv/CMakeLists.txt:10 (FIND_PACKAGE):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:123 (include)
```

```bash
dpkg-deb: error: parsing file '.dpkg_pkgs/jderobot-deps-dev/DEBIAN/control' near line 8 package 'jderobot-deps-dev':
 'Depends' field, missing package name, or garbage where package name expected
cp: cannot stat ‘.dpkg_pkgs/jderobot-deps-dev.deb’: No such file or directory
```

varhub, 2016-01-18:
First error is pretty simple: lack of **FindOpenCV.cmake**.
Why? because OpenCV or some cmake-common must provide it but it is not installed or debian package lack it.
**Solution**: look for it at internet and place it at system CMake directory (belong rest FindX).
*PS: also expecting missing or bad formatted opencv.pc*

Second error was always here since #255 (jenkins log [160](http://jenkins.jderobot.org/job/JdeRobot-GitHub/160/console))
```
 dc9cd7c       2015-12-15       Merge pull request #255 from jderobot-varribas/enhance/metapackages (Francisco Pérez)
```
Rationale: dpkg language processor found *missing* package at *Depends*. Therefore, there are two commas  or a bad package name.
Why? @fqez neither me can reproduce it. Only someone with access to jenkins machine can debug it.

**PS: please, edit message to provide links to jenkins and console log because I can't find above output.**
```
http://jenkins.jderobot.org/job/JdeRobot-GitHub/171/console
 > git checkout -f 1603395e67f0ff8dcd053065f3a0d178b642d3e0
FATAL: Could not checkout null with start point 1603395e67f0ff8dcd053065f3a0d178b642d3e0
stderr: fatal: cannot create directory at 'src/stable/components/gazeboserver/models/ArDrone/meshes/thermaleye_camera': No queda espacio en el dispositivo
```

varhub, 2016-01-18:
You should also check this error:
```
[JdeRobot-GitHub] $ /bin/sh -xe /tmp/hudson5910305704167931013.sh
+ rm -rf /chroot-dev/ubuntu-14.04/jenkins/ws/JdeRobot
+ mv /var/lib/jenkins/workspace/JdeRobot-GitHub /chroot-dev/ubuntu-14.04/jenkins/ws/JdeRobot
+ /var/lib/jenkins/scripts/jderobot-chroot.sh ubuntu-14.04
shell-init: error al obtener el directorio actual: getcwd: no se puede acceder a los directorios padre: No existe el fichero o el directorio
chdir: error al obtener el directorio actual: getcwd: no se puede acceder a los directorios padre: No existe el fichero o el directorio
```

closed 2021-02-21 by jmplaza