JdeRobot/base #313: libqwt5-qt4-dev not found in dev packages

almartinflorido, 2016-01-23:
Hi all,

I install the new version of JdeRobot in a clean virtual machine with Ubuntu 14.04 Gnome x86 and when I tried to install the dev packages (```jderobot-deps-dev```) I got the next error (or similar): ```libqwt5-qt4-dev not found```.

After of install the package with apt-get (```apt-get install libqwt5-qt4-dev```) I could install the dev package.

regards

fqez, 2016-01-23:
Yes, it's a well known problem that has no solution yet sadly. I will update the manual, so we install that package alongside ```jderobot-deps-dev```. Sorry for the inconvenience.

varhub, 2016-01-25:
This is an "ancient" issue documented since October 2015 http://jderobot.org/Manual-5#TIP:_libqwt5-qt4-dev_dependency_failure

Resolve it silently may lead in a future system instability due soft-inter-incompatibility between Qt packages.

almartinflorido, 2016-01-25:
Ok, thanks @fqez @varhub

closed 2016-01-25 by almartinflorido