JdeRobot/base #316: Gazebo execution errors

almartinflorido, 2016-01-25:
Hi,

when executed the world ardrone-turtlebot appears the next message in log file (```server/default.log```):

```
...
(1453746915 92445851) [Err] [ODEJoint.cc:495] SetParam(vel) boost any_cast error:boost::bad_any_cast: failed conversion using boost::any_cast
(1453746915 92528289) [Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
(1453746915 92610446) [Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
(1453746915 92698015) [Err] [ODEJoint.cc:495] SetParam(vel) boost any_cast error:boost::bad_any_cast: failed conversion using boost::any_cast
(1453746915 92780595) [Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
(1453746915 92862530) [Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
...
```

In a short execution with this world the log file grows extremely fast.

regards

fqez, 2016-01-26:
This is a known issue too, but sadly there is no easy solution for it. I wrote an email like a year ago to a Gazebo developer, and he told me that this warnings are there on purpose, so in order to get rid of them, you should modify the proper header in Gazebo source and recompile Gazebo.

almartinflorido, 2016-02-06:
For now we can set the file permissions to not writable.

Thanks @fqez

closed 2016-02-06 by almartinflorido