JdeRobot/base #318: The gazebo world "montmelo-pioneer" is not working

chanfr, 2016-01-27:
Hi all,
Im trying to run montmelo-pioneer and I getting the following message:

LaserDump Constructor
Load: 1496678224 cam_sensor_left
Load: 315815704 cam_sensor_right
 laser: pioneer2dx::laser
constructor Motors
Wheel Separation:0.280001
Wheel Diameter:0.222391
Laser Endpoints > default -h localhost -p 9996
constructor pose3dencoders
constructor Encoders
motors name pioneer2dx
Motors Endpoints > default -h localhost -p 9999
GetName() encoders: pioneer2dx
Pose3DEncoders1 Endpoints > default -h localhost -p 9993
Pose3DEncoders2 Endpoints > default -h localhost -p 9992
Pose3DMotors1 Endpoints > default -h localhost -p 9991
Pose3DMotors2 Endpoints > default -h localhost -p 9990
Encoders Endpoints > default -h localhost -p 9997
PropertiesI.cpp:407: Ice::FileException:
file exception: No such file or directory
path: cam_sensor_right.cfg
PropertiesI.cpp:407: Ice::FileException:
file exception: No such file or directory
path: cam_sensor_left.cfg
Conflicting gazebo versions

so,  cam_sensor_right.cfg and cam_sensor_left.cfg are not available under the configuration folder.
I followed the instruction on [1]. I copied the configuration file from:
/usr/local/share/jderobot/gazebo/worlds/
/usr/local/share/jderobot/gazebo/models
/usr/local/share/jderobot/gazebo/plugins/*/*cfg

And also from:
/usr/local/share/jderobot/conf/

thanks in advance,
Fran. 

[1]http://jderobot.org/Teaching_robotics_with_JdeRobot

fqez, 2016-02-03:
Hi Fran,

it works for me. Your problem is the name of the configuration file. Some time ago, pioneer were refactored so we could distinguish the configuration file of its cameras from other robots (like turtlebot). This removes some conflics until we adopt a new naming convention (as proposed at #225).

So in order to make the world work, you only have to rename the cam_sensor_{left,right} to cam_pioneer_{left,right}. In the case of turtlebot you should name it as cam_turtlebot_{left,right}

Hope it helps!

fqez, 2016-02-03:
Another option (seems more correct to me) is delete the redefinition of pioneer in the .world file and include it from the existing models. So instead of

```
<model name='pioneer2dx'>
...
</model>
```

you put:

```
<include>
      <uri>model://pioneer2dxJde</uri>
      <pose>0 0 0 0 0 0</pose>
</include>
```

and it should work. I will make a PR solving this!

chanfr, 2016-03-01:
It works thanks.

closed 2016-03-01 by chanfr