JdeRobot/base #328: Example of simulated ArDrone and uav-viewer does not work, new drone plugin

jmplaza, 2016-02-06:
The directions to launch this example are [http://jderobot.org/index.php/Running_JdeRobot#Simulated_ArDrone_.2B_UAVViewer here]
It seems a problem with the world configuration file 

Quadrotor2 is the new plugin (developed by @varhub) for the simulated ArDrone (#229). The gazebo world files must be changed in order to launch this plugin instead of the old one.

fqez, 2016-02-10:
merged at #330 Thanks!

jmplaza, 2016-02-27:
Still is broken. 
(a) The ArDrone.world in package is outdated, it does not use the ArDrone2 model. This file has been updated in the git repository but not in the package.
(b) Gazebo does not locate the ArDrone2 model. That model must be copied from the directory where the package installs it to ~/.gazebo/models and gazebo launched from there. Or the environment variables of Gazebo must be changed to include the /usr/local/share/jderobot/gazebo/models directory, in order to search for models in that directory too.
(c) the contents of /usr/local/share/jderobot/conf/uav_viewer_simulated.cfg does not match with those of ArDrone.world gazebo configuration file

fqez, 2016-02-28:
Should be fixed with the new package. v5.3.1-rc4

jmplaza, 2016-02-28:
1.- It works when gazebo is launched with the absolute path of its world file:

gazebo /usr/local/share/jderobot/gazebo/worlds/ArDrone.world

But still silently fails when launched with the world filename alone:

gazebo ArDrone.world

So it seems that gazebo does not find the world file on its search directories. It can be solved extending the GAZEBO_RESOURCE_PATH environment variable from:

export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-5.1:${HOME}/.gazebo/models:/usr/local/share/jderobot/gazebo/models

to 

export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-5.1:${HOME}/.gazebo/models:/usr/local/share/jderobot/gazebo/models:/usr/local/share/jderobot/gazebo/worlds

2.- Now it successfully uses the ArDrone2 plugin (from @varhub).

fqez, 2016-03-03:
Merged... again!

closed 2016-03-03 by fqez