JdeRobot/base #329: Example of cameraserver+cameraview revisited, libEasyIceConfig

jmplaza, 2016-02-06:
Yesterday, the example of cameraserver+cameraview (directions to launch it are here http://jderobot.org/index.php/Running_JdeRobot#Cameraserver_.2B_Cameraview ) didn't work for two students. It failed to locate cameraserver.cfg, cameraviewer.cfg and cameraviewer.glade files.

The files are already installed by the jderobot package at the corresponding directories but they are not found when running the components from other directories. libEasyIceConfig was designed to solve this topic, searching for this glade and configuration files at certain directories. Some more testing on this is needed.

fqez, 2016-02-10:
This doesn't work because the current pacakage was generated before merging the easyice and resourcelocator integration in each jderobot component. The next package will solve this issue :)
It should work now if the source code is compiled!

fqez, 2016-02-28:
Fixed with the new package v5.3.1-rc4

jmplaza, 2016-02-28:
It works like a charm with the new release, just following the instructions in the manual as they are. Good job!

closed 2016-03-04 by fqez