JdeRobot/base #337: Absolute interfaces for Motors ICE interface

jmplaza, 2016-02-11:
Motors ICE interface is more easily usable in different robots if it contains absolute commands for speed, in International System of Units: (m/s) for translation speeds and (rad/s) for rotation speeds. We have to check/adopt this in our KobukiDriver (and corresponding Gazebo plugin for simulated Kobuki), KobukiViewer, etc...

lr-morales, 2016-02-11:
Unifying the Motors interface usage is a good idea, not only for the Kobuki related drivers, plugins and tools but also for the generic control tools. 

Using absolute values with normalized units in is a good system, but maybe could pose a problem for generic control tools like teleoperator: each robot could have different maximum speeds, making "controls" like teleoperator (which relies on a joystick-like GUI to make the robot move) difficult to use without knowing beforehand the specifications of the final robot.

Maybe  this falls out this issue, but I think we should think how to handle this somehow if we want to maintain these kind of controls (i.e. via config file, options, specials value to interface...).

jmplaza, 2016-02-11:
yes :-) We will have to adapt the teleoperators to use these absolute units in the Ice Interface.

We can put absolute values in the GUI of the teleoperator, or even better, keep the GUI without any number but allow specifying different maximum speed values in the configuration file of the teleoperator. The teleoperator may have default MaxSpeed values valid for the most commonly used robot. Those values work even when the teleoperator configuration file does not include any MaxSpeed parameters. And when the teleoperator is used with a different robot, the right values for its MaxTraslationSpeed and MaxRotationSpeed must be specified in the teleoperator configuration file.

But all these transformations must be dealt in the client side, not in the driver. The driver cleanly understand commands in International System of Units.

fqez, 2016-03-03:
First approach at #358 I've measured the units per second of both linear and angular velocity and it seems to be almost the right units (please someone else test it). There is no MaxSpeed configuration throught .cfg files yet because it was giving me a headache, but it will be soon.

For testing:
  * Clone the code at [absolute_interfaces](/fqez/JdeRobot/tree/absolute_interfaces)
  * Custom compilation with: ```cmake . -Dbuild-default=OFF -Dbuild_kobukiViewer=ON -Dbuild_gazeboserver=ON -Dbuild_turtlebot=ON ```
  * source gazebo setup: ```source gazebo-setup.sh```
  * Launch gazebo and insert a turtlebot model
  * Launch kobukiViewer and test the behaviour.

-->[Here's](http://jderobot.org/store/fperez/uploads/videos/absolute_interfaces.webm) <-- a little video showing how it works. I've include a tooltip on the interface that shows the values of v and w at real time.

closed 2016-03-17 by fqez