JdeRobot/base #340: the package generation for Debian is broken

jmplaza, 2016-02-18:
Right now only packages for Ubuntu 14.04 are generated (both 32 and 64 bits)

lr-morales, 2016-05-12:
While trying to install and manually fix the generated packages, I have met these issues:
- Some CMAKE_CXX_FLAGS redefinitions break compilation for gcc-1.9.2 in Debian Jessie. This problem is addresed in the issues #435 and #436. 
- Packages are generated with wrong-named duplicated dependences in Debian Jessie, probably related to the DEPS variable population commented in #252; i. e. upstream Jessie pcl library names follow the scheme libpcl-[Module]1.7, but the ubuntu and DEPS libraries scheme is libpcl-[Module]-1.7. This results in unmet dependencies.
- Some modules are built and in their locations in the build path, but aren't transferred to the final package.

fqez, 2016-07-25:
They now can be made without trouble :)

closed 2016-07-25 by fqez