JdeRobot/base #343: KobukiViewer not found in JdeRobot installation from packages

almartinflorido, 2016-02-23:
Hi,

I installed jderobot and jderobot-teachingrobotics packages and the component kobukiViewer and kobukiViewer's configuration file are missing.

Ubuntu 14.04 x64

regards

jmplaza, 2016-02-27:
Not only that. The example [Simulated Kobuki + KobukiViewer](http://jderobot.org/index.php/Running_JdeRobot#Simulated_Kobuki_.2B_KobukiViewer) is broken!.

(a) kobukiViewer component and the kobukiViewer.cfg are in the repository but they are not installed when the jderobot package is installed. 
(b) turtlebot.world file mentioned in the example is not in the repository neither in the jderobot github. It is better to name it kobuki.world.

fqez, 2016-02-28:
Should be fixed with the brand new package (v5.3.1-rc4). I've just uploaded it!

jmplaza, 2016-02-28:
Uhm... still does not work. The (a) part it is solved: kobukiViewer is installed and its default configuration file too. 

But (b) part still does not work: the 'turtlebot.world' for launching Gazebo is missing (both from the installedo package and from the github repository), so Gazebo fails to launch the nice office environment shown in the video. For better name convention such world file should be named 'kobuki-simple.world' or something like that, and included in the git repository and in the next JdeRobot package.

[image]

closed 2016-03-09 by fqez