JdeRobot/base #382: Problem while loading F1 SimpleCircuit

chanfr, 2016-03-15:
Hi,
the related world seems to work fine but I get an error while trying to load the world that makes gazebo to take too much time to start:

(1457979550 482804598) [Wrn] [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
(1457979592 149712701) [Err] [ModelDatabase.cc:408] Unable to download model[model://f1/meshes/hokuyo.dae]
(1457979592 149762373) [Err] [SystemPaths.cc:399] File or path does not exist[""]
(1457979592 149791750) [Err] [Visual.cc:2360] No mesh specified
(1457979592 201650409) [Err] [Material.cc:93] Unable to find texture[Color_006.jpg] in path[/home/frivas/.gazebo/models/pistaSimple/meshes]
(1457979668 156534189) [Wrn] [Publisher.cc:131] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.

any idea?

fqez, 2016-03-15:
Yes, my fault. For some reason the custom hokuyo laser mesh wasn't included in the f1 model pull request. I will fix this ASAP.

Sorry!

fqez, 2016-03-15:
Done, should be working now. Let me know if you face some other problems.

closed 2016-03-31 by fqez