JdeRobot/base #386: F1 car model needs better dynamics

jmplaza, 2016-03-16:
the current ones are not realistic

fqez, 2016-03-18:
Merged!

fqez, 2017-03-24:
The new plugin improves the previous one in the following aspects:

* The main enhancemente is the movement style. The current plugin moves the model as a whole; that means that the wheels are not making the car moves, but it moves like a box. The new plugin has two steering wheels and two driving ones, so the car moves by the friction generated between the driving wheels and the ground. This allows us to have some inertia in the movement, so the simulation is more realistic.
The acceleration and the steering is controlled by two PID controllers, so we can simulate the resistence (friction) and the inertia when the car throtles or brakes.

* Also, now the plugin connects all its interfaces to an unique port, making the configuration file easier to type and configure. This is inherited from quadrotor2 and turtlebot plugins.

There is some work to do yet, like the control of the steering wheels when the car stops suddenly (hand brake), because now it drifts in a random direction when this occurs. Also, the limits of the steering wheels rotation needs to be optimized, it is currently very low (almost like a real F1 car).

commit 8e886f86 (2017-05-05): Merge pull request #736 from fqez/issue_386_2

[fix #386] Added new Formula1 plugin
closed 2017-04-06 by fqez