JdeRobot/base #390: Wrong detection of the pioneer color in the ardrone-turtle.world

reysam93, 2016-03-22:
When launching the world ardrone-turtlebot.world, the pioneer is represented with the top green and blue, but when I start using the drone's camera, the camera do not detect this colors. There is attached an image of how it looks. I have tried it using the turtlebotJde2cam, the turtlebotJde, the Ardrone2 and the Ardrone, but non of them works.

[image]

fqez, 2016-03-26:
Try the new world ardrone-turtlebot_transparent.world. Should solve your problem. Remember to copy the new model ground_plane_transparent too to ~/.gazebo/models

Cheers

reysam93, 2016-03-31:
I have tried it and still does not work properly. As you can see in the picture, looks like the green zone over the robot is transparent, the camera can only see the borders.
[image]

fqez, 2016-03-31:
It seems like the model is colliding with the tag as happened with the ground. We have two options here, make the turtlebot model transparent too, or review the tag. I will take a look to the tag and play with different textures.

artuvd, 2016-07-25:
I have the same problem! I've tried ardrone-turtlebot_transparent.world too, with the same result that reysam93. It seems like the tag is transparent under the bot is, in other tag parts the camera detects it.

fqez, 2016-07-25:
Yes, I tried multiple solutions here, but no one seems to work. I think is a render issue from the gazebo cameras, but it's not characterized yet. We will update our dependencies to gazebo 7 so we can test there if this issue is solved. If not, I will give another try to this problem. 

Sorry for the inconvenience.

fqez, 2016-07-26:
I've tried again in a Ubuntu 14.04 64-bit native machine and works for me and for @aitormf. Are you using a virtual machine?

Here there is two examples testing this issue

* With uav_viewer

[image]

* With introrob_py

[image]

both of them running on native systems.

artuvd, 2016-07-26:
I had before a virtual machine (in Parallels) and it worked perfectly. Now i've prepared a physical machine, a HP-Compaq CQ-70, running ubuntu 14.04 LTS, and it hasn't any other OS installed, it's dedicated exclusively to this proyect.

aitormf, 2017-05-12:
Close this issue because actually works well: 

[image]

closed 2017-05-12 by aitormf