JdeRobot/base #391: Replayer can only be used with one interface at a time

samartin, 2016-03-23:
The configuration file of replayer component should allow to configure different endpoints for connecting to more than one interface. It also should be included in the documentation.

fqez, 2016-04-25:
Hi Samuel,

it CAN be used with more than one interface at the same time, but it's pretty tricky to configure properly. I'm uploading a video showing how to configure it and make it works with the camera and laser interfaces at the same time. In 10 min or so you will have it ;D

fqez, 2016-04-25:
Here you are: http://jderobot.org/store/fperez/uploads/videos/RR-config-computer.webm

samartin, 2016-05-01:
Thank you so much for the tutorial!

My problem was that I thought I needed to connect to more than one port. I was able to replay a camera and connect it to cameraview, for example, but I did not know I had to use the same port to connect another interface.

Since it is not an issue I close it.

closed 2016-05-01 by samartin