JdeRobot/base #396: gazebo does not find its resources from the JdeRobot directories

jmplaza, 2016-03-28:
After installing JdeRobot Gazebo should find all the needed resources (models, worlds, plugins...) from the directories where JdeRobot package installs them. This will avoid the need for a post-install script that copies all the resources in the local HOME/.gazebo directory. Gazebo should be configured to do so. Now the file [https://github.com/RoboticsURJC/JdeRobot/blob/master/src/stable/components/gazeboserver/gazebo-setup.sh] takes care of this, but it seems to work only partially.

closed 2016-03-31 by fqez

varhub, 2016-04-01:
**Bug remains**
Tested latest package (rc7) and TeachingRobotics.
Just same error as https://github.com/RoboticsURJC/JdeRobot-TeachingRobotics/issues/41

#393 is a **must to do**

```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.1.54.167
[Wrn] [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
[Err] [ModelDatabase.cc:408] Unable to download model[model://pistaSimple]
[Err] [SystemPaths.cc:399] File or path does not exist[""]
Error [parser.cc:528] Unable to find uri[model://pistaSimple]
[Err] [ModelDatabase.cc:408] Unable to download model[model://f1]
[Err] [SystemPaths.cc:399] File or path does not exist[""]
Error [parser.cc:528] Unable to find uri[model://f1]
```

varhub, 2016-04-01:
Forget it!, gazebo-setup not sourced.
#393 is required just for this, but .bashrc injection must be realized standalone.

fqez, 2016-04-01:
LOL! sourcing gazebo-setup.sh it works fine? Just to know!

Thanks for test it! :D

varhub, 2016-04-01:
For future testing refer to:
https://github.com/jderobot-varribas/jderobot-post-install/blob/master/jderobot_super_installer.sh