JdeRobot/base #402: 3DViewer does not compile

fqez, 2016-03-31:
Hi,
I'm having an issue with the new 3DViewer component. It fails when trying to compile. This is the output:

```
Scanning dependencies of target 3DViewer
[ 48%] Building CXX object src/stable/components/3DViewer/CMakeFiles/3DViewer.dir/main.cpp.o
[ 48%] Linking CXX executable 3DViewer
/usr/bin/ld: ../../libs/jderobotViewer/libjderobotViewer.a(JderobotViewer.cpp.o): undefined reference to symbol 'sem_post@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/stable/components/3DViewer/3DViewer] Error 1
make[1]: *** [src/stable/components/3DViewer/CMakeFiles/3DViewer.dir/all] Error 2
make: *** [all] Error 2
```

Any clue?

chanfr, 2016-03-31:
The compilation problem can be due to #401. I fixed it, can you check it again?

fqez, 2016-03-31:
Nope, I did that change before, but it still does not compile. :S

chanfr, 2016-04-01:
I've just compile in another computer and it worked... 

It seems to be a problem while linking with pthread, can you force to include it into the 3DViewer CMakeLists.txt?

        TARGET_LINK_LIBRARIES(3DViewer
                        jderobotutil
                        ${OpenCV_LIBRARIES}
                        ${OpenCVGUI_LIBRARIES}
                        colorspacesmm
                        JderobotInterfaces
                        ${easyiceconfig_LIBRARIES}
                        ${ZeroCIce_LIBRARIES}
                        ${CMAKE_THREAD_LIBS_INIT}
                        ${ZLIB_LIBRARIES}
                        logger
                        crypto
                        ns
                        jderobotViewer
                        ${PCL_LIBRARIES}
                        pthread
        )

fqez, 2016-04-01:
Yep it seems there is some kind of problem with my VM and pthread. Thanks!

closed 2016-04-01 by fqez

abhijayghildyal, 2019-03-10:
I'm facing the following error:

	Scanning dependencies of target 3DViewer
	[ 55%] Building CXX object src/tools/3DViewer/CMakeFiles/3DViewer.dir/main.cpp.o
	[ 55%] Linking CXX executable 3DViewer
	../../libs/jderobotViewer/libjderobotViewer.a(JderobotViewer.cpp.o): In function `jderobot::JderobotViewer::getScreenshotImage(cv::Mat&)':
	JderobotViewer.cpp:(.text+0x4c6b): undefined reference to `cv::imread(cv::String const&, int)'
	../../libs/jderobotViewer/libjderobotViewer.a(JderobotViewer.cpp.o): In function `cv::String::~String()':
	JderobotViewer.cpp:(.text._ZN2cv6StringD2Ev[_ZN2cv6StringD5Ev]+0x14): undefined reference to `cv::String::deallocate()'
	../../libs/jderobotViewer/libjderobotViewer.a(JderobotViewer.cpp.o): In function `cv::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
	JderobotViewer.cpp:(.text._ZN2cv6StringC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2cv6StringC5ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x70): undefined reference to `cv::String::allocate(unsigned long)'
	collect2: error: ld returned 1 exit status
	src/tools/3DViewer/CMakeFiles/3DViewer.dir/build.make:407: recipe for target 'src/tools/3DViewer/3DViewer' failed
	make[2]: *** [src/tools/3DViewer/3DViewer] Error 1
	CMakeFiles/Makefile2:3851: recipe for target 'src/tools/3DViewer/CMakeFiles/3DViewer.dir/all' failed
	make[1]: *** [src/tools/3DViewer/CMakeFiles/3DViewer.dir/all] Error 2
	Makefile:149: recipe for target 'all' failed
	make: *** [all] Error 2

What do you think might be the problem?

abhijayghildyal, 2019-03-10:
I fixed it by installing opencv from source

https://github.com/anujonthemove/opencv3-installation

jmplaza, 2019-03-11:
Hi @abhijayghildyal , the 3DViewer has been moved to the [Viz repo](https://github.com/JdeRobot/viz) and renamed as 3DVizWeb