JdeRobot/base #409: ColorTuner core dump

varhub, 2016-04-17:
ColorTuner presents random startup coredumps.
It could be related to monitor resolution.

Reproduced in second PC.

Ice connection error is almost discarded.

ojgarciab, 2016-04-18:
What kind of exception are?

varhub, 2016-04-20:
No one. Just a core dump with no more hint.

ojgarciab, 2016-04-20:
I can't reproduce the core dumps, but I'm getting noise, random color changes (to purple and sometimes to red) in some frames (not always, once every 20-30 seconds). Maybe is related with some bad pointer, sharing some image mat or something like that.

It's first time to me with colorTuner, so I'll review the code to try to find some bug.

these were my attempts to crash the program:

```
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Conexión rehusada
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ cat colorTuner.cfg 
Cameraview.Camera.Proxy=cameraA:tcp -h localhost -p 9998
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ vi colorTuner.cfg 
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ cat colorTuner.cfg 
Cameraview.Camera.Proxy=cameraA:tcp -h localhost -p 9999
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
474,289 23 15 1
6.95308e-310 0 0 2.22787e-316 6.93074e-310 2.22787e-316 
391,204 27 21 23
6.95308e-310 0 0 2.228e-316 6.93074e-310 2.228e-316 
298,147 149 73 30
6.95308e-310 0 0 2.228e-316 6.93074e-310 2.228e-316 
348,408 98 51 27
6.95308e-310 0 0 2.22799e-316 6.93074e-310 2.22799e-316 
339,448 146 122 102
6.95308e-310 0 0 2.27455e-316 6.93074e-310 2.27455e-316 
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
redstar@nvidiastar:~/Documentos/github/JdeRobot/src/stable/components/colorTuner$ ./colorTuner --Ice.Config=colorTuner.cfg
Loading interface...correct
[Info] loaded Ice.Config file: colorTuner.cfg
279,264 102 58 15
6.95328e-310 0 0 9.71726e-317 6.95238e-310 9.71726e-317 
257,234 153 70 29
6.95328e-310 0 0 9.68667e-317 6.95238e-310 9.68667e-317 
147,205 176 125 69
6.95328e-310 0 0 9.23536e-317 6.95238e-310 9.23536e-317 
```

ojgarciab, 2016-04-20:
Umm, I just suffer the problem.

I had cameraserver configured at 5 fps, I reverted the configuration file to 25 fps and begin to get coredumps 3 of 4 executions.

I'm going to do a prior quick review of code.

varhub, 2016-04-20:
Oh! these are good news. FPS related.
That will get worse with virtual machine or under stress.

ojgarciab, 2016-04-20:
I'm trying to reproduce again the core dumps and I'm not lucky :'(

Please, take a look to this signal catcher that I developed to debug the problem:
https://github.com/ojgarciab/JdeRobot/commit/ef59314885e3aa7720f3964aa2c8e32b44535d69#diff-a5198cb912bd1b95901c4239701353c3

```
#include <signal.h>
#include <unistd.h>
#include <execinfo.h>

void handler(int signal) {
    void *trazas[20];
    size_t tam;
    fprintf(stderr, "Señal capturada: %d\n", signal);
    tam = backtrace(trazas, 20);
    backtrace_symbols_fd(trazas, tam, STDERR_FILENO);
    _exit(1);
}

int main(int argc, char** argv){
    int status;

    signal(SIGSEGV, handler);   // install our handler
```

I have to go to sleep now, please, try the code. See tomorrow!

commit f2a82243 (2017-03-01): Merge pull request #697 from aitormf/issues/692

[issue #692] Deleted colorTuner c++
fix #692, fix #238, fix #246, fix #409, fix #416
closed 2017-03-01 by fqez