JdeRobot/base #415: compilation of the driver 'ardroneserver' from source code seems to be broken

jmplaza, 2016-04-20:
This driver provides support for real ArDrone 2.0 quadrotor from Parrot. The compilation chain of JdeRobot from source seems to be broken with this driver as @mazafrav pointed out [http://jderobot-developer-list.2315034.n4.nabble.com/JdeRobot-Error-in-compilation-td4642992.html#a4642996]. This driver depends from ardronelib which we uploaded to our ThirdParty repository [https://github.com/RoboticsURJC/JdeRobot-ThirdParty/tree/master/ardronelib]. 

The compilation chain should be reviewed. It is already fixed somewhere as the debian package of JdeRobot is properly generated from the GitHub code, including this driver. Maybe the solution is as simple as installing first the ardronelib from our ThirdParty repository.

fqez, 2016-04-20:
It works in the chroots and jenkins does not complain about it. Also I do a daily compilation of the whole project for testing, and works as well.. this issue may be an isolated problem from @mazafrav I guess. I couldn't reproduce it in 3 different machines. My suggestion for him is to reistall the packages or try a manual installation, to fix dependencies.

mazafrav, 2016-04-22:
I solved it, it was an isolated problem indeed. As Victor (@varhub) pointed in the correspondent list thread i was lacking the c++11 flag, so i switched to g++4.8 (i was using 4.7) and everything worked fine.

fqez, 2016-04-25:
Nice! thanks @varhub

closed 2016-04-25 by fqez

varhub, 2016-04-28:
Yep. By the way, just for historical purposes, error is due leak of c++0x (2008-09, not 2011).
I'm doing struct = {what, ever} since I began career ;)

> Going back to initial error "Pose3DI::xyzT = {0, 0, 0}", I think you lack c++0 or c++11 flag/support.
> Since c++0 you can initialize structs with {}
> This kind on complain will happen at c++99