JdeRobot/base #420: portability: test of 'Simulated ArDrone + introrob.py' from Windows

jmplaza, 2016-04-25:
Similar to #419 but with a Pyhton component instead of a C++ one. 

One attractive feature is portability, to use JdeRobot from other Operating Systems. Docker may help on this. Let's do a test running part of the 'Simulated ArDrone + introrob.py' example in TeachingRobotics [http://jderobot.org/Teaching_robotics_with_JdeRobot#Simulated_ArDrone_.2B_introrob_py] on a Windows machine. Gazebo will run on a Linux machine as usual, but the introrob.py component should run on a Windows machine.

aitormf, 2016-08-02:
It is not exactly the same, but I could use the exercise Teaching Robotics of FollowLines on Windows, running gazebo in a [Docker ](http://www.docker.com/)(comes to be similar to a virtual machine).
You can use the same dependencies versions that in Ubuntu (Python 2.7, PyQt4, Qt4.8.7,...) excepts **OpenCV** and **Ice**.

- OpenCV: I saw the 3.1 versión to Python and when I tried it, Works without any code change.
- Ice: I use 3.6 versión because I could not run 3.5.  It works without any code change.

[image]

I decided start by this instead of Introboy_py because firstly I had problems with a dependencie of introrob_py (PyQwt to flying Indicators) and because FollowLines only has 2 Ice Interfaces (camera and motors).

I needed change EasyIceConfig library because it has harcoded paths that no Works in Windows. And now I'm doing Parallelice_py library because the architecture of TeachingRobotic clients does that their GUI to hang while the sensor data are being received from the server.

[image]

**Instructions to install all depedencies in Windows:**

1. Download python 2.7.12 and install checked env variables (Is posible that you need restart to run the PATH): [https://www.python.org/ftp/python/2.7.12/python-2.7.12.amd64.msi](https://www.python.org/ftp/python/2.7.12/python-2.7.12.amd64.msi)

2. Download qt 4.8.7: [http://download.qt.io/official_releases/qt/4.8/4.8.7/qt-opensource-windows-x86-vs2010-4.8.7.exe
](http://download.qt.io/official_releases/qt/4.8/4.8.7/qt-opensource-windows-x86-vs2010-4.8.7.exe)

3. update pip:
<pre>python -m pip install --upgrade pip</pre>

4. Install deps:
<pre>pip install numpy zeroc-ice
pip install http://jderobot.org/store/aitormf/uploads/windows/opencv_python-3.1.0-cp27-cp27m-win_amd64.whl
pip install http://jderobot.org/store/aitormf/uploads/windows/PyQt4-4.11.4-cp27-none-win_amd64.whl
pip install http://jderobot.org/store/aitormf/uploads/windows/PyQwt-5.2.1-cp27-none-win_amd64.whl
</pre>

5. Install jderobot libraries and unzip in c:\python27\Lib:  [http://jderobot.org/store/aitormf/uploads/windows/jderobot_cp27_win64.zip](http://jderobot.org/store/aitormf/uploads/windows/jderobot_cp27_win64.zip)

6. Download TeachingRobotics and run from CMD

aitormf, 2016-08-16:
Now we can use introrob py + simulated Ardrone in Windows (video):
[image]

aitormf, 2016-09-28:
this issue is resolved with #543

closed 2016-10-24 by fqez