JdeRobot/base #425: Missing library ICE-JS in JdeRobot dependencies

aitormf, 2016-04-28:
Missing library ICE-JS in JdeRobot dependencies. It allows servers to use the WebSocket protocol required for web clients. Currently cannot be used without first going to my repository to download it.

Tests that do not come into conflict with any JdeRobot component should be made

fqez, 2016-05-16:
Which exact library is missing?

aitormf, 2016-05-17:
To use the webTools, need to install the IceWS (from IceJS) plugin for servers ICE can use the WebSocket Protocol. To do this we need to follow the instructions I have on my [MediaWiki](http://jderobot.org/Aitormf-tfg#IceJS) but I think that this installation should be within the premises of JdeRobot.

closed 2016-10-24 by fqez