JdeRobot/base #440: [drivers] support for Kinect1 sensor is broken

jmplaza, 2016-05-08:
Openniserver works like a charm with ASUS Xtion sensors. We tried openni1server for Microsoft Kinect sensors and it does not work. It seems to connect at low level:

jderobot@theta02:~$ lsusb
Bus 002 Device 011: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 002 Device 007: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 002 Device 009: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 002 Device 005: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 002 Device 010: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 002 Device 006: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 002 Device 008: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 002 Device 002: ID 0409:005a NEC Corp. HighSpeed Hub

but it fails to detect the RGBD devices:

NCAMERAS = 2
Number of detected devices: 0
Enumerate failed: Can't create any node of the requested type!
Creating camera cameraA
openni1Server.CameraRGB.PlayerDetection
Starting thread for camera: cameraA
              -------- openni1Server: Component: CameraRGB created successfully   --------
Creating camera cameraB
Starting thread for camera: cameraB
              -------- openni1Server: Component: CameraDEPTH created successfully   --------
RGB FPS: 10
WaitAnyUpdateAll() failed failed: A timeout has occurred when waiting for new data!

Is Kinect-1 supported by openniserver? I thought we kept the openni1server just because it included the support for this sensor.

chanfr, 2016-05-09:
I think openni1Server should be label as deprecated as kinect1 sensor has support on openni2. The only "trick" is that you have to compile an external driver (not included with openni2). The instructions can be found here:
http://jderobot.org/Frivas-tfm#Kinect_working_with_openni2

Anyway I will confirm that this is currently working this afternoon.

chanfr, 2016-05-11:
Here is the new repo for the openni2 kinect driver:
https://github.com/RoboticsURJC/libfreenect/tree/master/OpenNI2-FreenectDriver

fqez, 2016-05-16:
After some testing, I'm not able to connect the device to openniServer. I've tried to compile OpenNI2 from the official repo, and libfreenect works fine with NiViewer provided by openni. But emulating the same steps for openniServer does not make any difference. Any ideas?

closed 2017-04-06 by fqez