JdeRobot/base #441: [tools] recorder-replayer out of sync

jmplaza, 2016-05-09:
As pointed out by @samartin in #202 there is a relevant delay and out of sync when replaying several sources of sensor data stored in log file.

samartin, 2016-05-09:
In [this test](http://jderobot.org/store/samartin/uploads/videos/RecorderReplayerSyncProblem.og) a delay can be seen between the images and the position. The images arrives first. As I said in the other issue, I move the camera once and then wait 2 o 3 seconds. In [this other test](http://jderobot.org/store/samartin/uploads/videos/RecorderReplayerLaserCamera.ogv), I used a laser and a camera interface, and It seems like there is no delay between them. So it can be a problem of some interfaces or just pose3d since it is the new one.

samartin, 2016-06-21:
The problem is this [line](https://github.com/RoboticsURJC/JdeRobot/blob/master/src/stable/tools/replayer2/replayer.cpp#L862). It adds 1 extra second in the Pose3D thread so its update rate is 1 Hz approximately. This also affects to the Pose3DEncoders thread. 

In [this video](http://jderobot.org/store/samartin/uploads/videos/ReplayerFixed.ogv) I show a test of the fix.

fqez, 2016-06-22:
Merged, thanks @samartin

closed 2016-06-22 by fqez