JdeRobot/base #443: [tools] rgbdmanualcalibrator does not have an entry in the web manual yet

jmplaza, 2016-05-09:
so there is very little information about how to use it. Now we are developing more applications with RGBD sensors and the use of this tool is growing.

fqez, 2016-05-09:
It has indeed an entry in the tools section of the manual http://jderobot.org/index.php/Tools#CameraCalibrator It only miss the video, it does not exist anymore for some reason.

jmplaza, 2016-05-09:
Uhm... that one is for CameraCalibrator, a different tool. We have three different camera calibrators: (a) CameraCalibrator (based on OpenCV and a cheesboard pattern), (b) rgbdmanualcalibrator (that allows the user to manually change the calibration parameters --with sliders in the GUI-- and see the world --defined as lines in a configuration file-- proyected into the images according to those calibration parameters), and (c) rgbdcalibrator (allows a semiautomatic calibration of the rgbdcamera using a 3D pattern). All of them are poorly documented in the web manual. Let's go step by step improving this lack of good documentation. Let's start with rgbdmanualcalibrator. 

By the way all of them should be moved to tools directory.

fqez, 2016-05-09:
Oh ups! yeah I missread, sorry. You are right!

closed 2021-02-21 by jmplaza