JdeRobot/base #457: [drivers] Quadrotor2 plugin now depends on Quadrotor plugin

jmplaza, 2016-05-23:
and should be self-contained. Quadrotor old plugin will be replaced by the new (better) release and deleted from the repository.

aitormf, 2016-06-30:
I do not see this dependence, I've been looking at the code and  CMakelist.txt and it does nothing. It may seem by how the files are sorted but not depends on quadrotor. And you can compile quadrotor2 without quadrotor

almartinflorido, 2016-06-30:
Hi @aitormf ,

I think that @jmplaza  refers to this [1], the meshes and materials of ArDrone2 referenced to ArDrone. The idea is remove ArDrone model and plugin. Furthermore, the model ArDrone has meshes do not correspond with the quadrotor [2].

regards

[1]: https://github.com/RoboticsURJC/JdeRobot/blob/master/src/stable/drivers/gazeboserver/models/ArDrone2/model.sdf
[2]:https://github.com/RoboticsURJC/JdeRobot/tree/master/src/stable/drivers/gazeboserver/models/ArDrone/meshes

closed 2016-07-22 by fqez