JdeRobot/base #458: Recorder GUI don't load in Linux package

aitormf, 2016-05-24:
I try run recorder with Linux package installation and it shows ths error:

<pre>Create consumer 
Create consumer 
Loading glade 
 
 
(recorder:3695): libglade-WARNING **: could not find glade file './recordergui.glade' 
terminate called after throwing an instance of 'Gnome::Glade::XmlError' 
Abortado (core generado) </pre>

I've seen it is because when loading the file .glade does:
`refXml = Gnome::Glade::Xml::create("./recordergui.glade"); `

Instead of (like cameraview):

<pre>const std::string gladepath = resourcelocator::findGladeFile("recordergui.glade"); 
… 
…
 refXml = Gnome::Glade::Xml::create(gladepath);</pre>

aitormf, 2016-05-24:
Also I've seen replayer has the same problem

fqez, 2016-05-24:
Hi! yes, recorder and replayer seems not to have the resource locator feature implemented, also recorder2 and replayer2 lack tihs functionality. It's easy to solve though.

fqez, 2016-06-10:
Thanks! merged

closed 2016-06-10 by fqez