JdeRobot/base #462: VisualHFSM: Defining proxy names

reysam93, 2016-05-25:
When defining the .cfg file, the name given should be the name that users can use for calling their proxy's interface, instead of using this name ended in Prx, which is how it works now.

fqez, 2016-06-10:
It is related with issue #459, isn't it? Let me know and I will close this issue as well, due to this changes are already merged.

reysam93, 2016-06-13:
They are different issues. Issue #459 refers to the proxy name for the ICE configuration file, this issue refers to the name of the object that the developer can use inside the code, because now this object's name are NamePrx, and it would be easier calling just Name, because if Name is the name the user gives them through the graphical editor it will be easier to remember.

fqez, 2017-06-13:
merged!

closed 2017-06-13 by fqez