JdeRobot/base #478: Failure to generate correct packages after a cmake + make single pass.

lr-morales, 2016-06-03:
Assumming that everything else is ready and clear for compilation, the command chain to make packages with all jderobot base code should be something like:
````
    mkdir build; cd build; cmake .. && make && make package
````

This or similar recipes seem to make incomplete packages with almost all binaries missing; this was checked against Debian Jessie x32 and x64 distributions.

The problem is Workarounded if CMake is executed again and then making the package. The problem seems to be the binary location check done in the CMakeLists.txt in order to issue the correct install commands:
````
`FOREACH (currentBin ${LIST_COMPONENTS})
    if (EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/src/stable/components/${currentBin}/$
{currentBin})
        INSTALL (PROGRAMS ${CMAKE_CURRENT_SOURCE_DIR}/src/stable/components/${cu
rrentBin}/${currentBin} DESTINATION /usr/local/bin OPTIONAL COMPONENT core)
    endif()
    if (EXISTS ${CMAKE_CURRENT_BINARY_DIR}/src/stable/components/${currentBin}/$
{currentBin})
        INSTALL (PROGRAMS ${CMAKE_CURRENT_BINARY_DIR}/src/stable/components/${cu
rrentBin}/${currentBin} DESTINATION /usr/local/bin OPTIONAL COMPONENT core)
    endif()
ENDFOREACH(currentBin)
````

The files don't exist in the first pass of cmake and they aren't "re-checked" when trying to run the install command.

fqez, 2016-06-10:
Happens as well in Ubuntu 14.04 x32 and x64. I need to double cmake in order to generate a proper package. It also compiles the whole project again even with the CMakeCache.txt file existing.... a true pain.

lr-morales, 2016-06-15:
Regarding this issue, it seems (by looking #279) that this check is made in order to support the installation of "existing binaries" and "compiled binaries".
The install commands as they are written will not issue any kind of error if the file couldn't be installed, so the questions are: 
* Is there any possible and _reasonable_ situation in which someone would have at the same time both binaries? (Where _reasonable_, could be perfectly a "situation which we want to be supported")

* If the previous answer was that _there is_, shouldn't the "existing one" prevail over the "compiled one"?

If there is no situation in which both binaries could exist, then we can remove the conditionals so cmake doesn't need a second pass. If one hipotetical existing binary could exist and should prevail the compiled one, the conditional could be changed to select between existing in source dir and compiled in binary dir.
I we don't know which one could be, I would go with the second option.

The last resort would be tweaking the install process with a custom script: it is possible to make CPack to register files installed from a custom script (tested on Debian x32) but isn't as straight-foward as an `install` command.

What do you think about this?

fqez, 2016-06-22:
I think cmake should need only one pass in all the cases. I can't think in a scenario in which we would need both binaries. So if removing the conditionals solve this issue, go ahead!

lr-morales, 2016-06-23:
I'm testing a fix which involves the binary check only once: if there is a binary in the source tree at config time, `install` will take that binary; the generated one will be used otherwise.

With this fix, the user should maintain the source tree clean of unwanted binaries (which is easy if uses out-source building).

When I have it tested and ready, I'll make the pull-request.

closed 2016-06-30 by fqez