JdeRobot/base #516: New organization of directory tree

fqez, 2016-07-05:
For the incoming release of JdeRobot (5.4.0) we are going to get rid of the following directories:

* stable
* testing
* components

So the new directory tree will be:

src/
   |_ drivers/
   |_ tools/
   |_ libs/
   |_ interfaces/

For this, we will need to check the implications on the root CMakeLists and the sub CMakeLists files as well. Also, it would be preferable to maintain the history of all the files.

fqez, 2016-07-05:
I've uploaded a debian package with this changes (JdeRobot 5.4.0) at the testing apt repository, so anyone can test it if he would.

fqez, 2016-07-05:
Also if @chanfr, @lr-morales and @aitormf can test it so I can make sure this works, great!
I've asked @varhub if there's a possibility about maintain the history of the files after the refactorization, but there is no way it seems. If anyone have more info about that I would appreciate it.

Thanks in advance!

aitormf, 2016-07-06:
Hi @fqez!!

I just try the ubuntu 64 bits packages and they have worked well

lr-morales, 2016-07-06:
Hello @fqez,
   I'm a little busy here due unforeseen circumstances, but regarding the file history, if git has identified correctly a file move/rename when commiting changes, ``git log --follow`` may track the file history beyond renaming (tries to "connect" the renamed detected files). As far as I read (which isn't too much, so possibly I'm wrong here), this "feature" isn't the default due the this type of tracking makes sense when doing a per file history log or something like that.

chanfr, 2016-07-08:
good job!

closed 2016-07-08 by chanfr