JdeRobot/base #524: Delete --Ice.Config flag in components

aitormf, 2016-07-21:
It is good idea to not have to use the flag --Ice.Config to add the configuration file of components. Just need to add the configuration file without flag

fqez, 2016-07-21:
Isn't it better to wait till we decide how to parse files first? So you can make the changes all together instead of refactor them twice!

aitormf, 2016-07-22:
I have found a way easy of do it and is modify the part of load the properties in easyconfig, so accept both flag as not. So we should only add easyconfig to components that do not have it

jmplaza, 2016-07-22:
If it is so easy to have this functionality, let's put it now to start enjoy it, no?. Launching components will be even more easy and intuitive than now.

closed 2016-08-31 by fqez