JdeRobot/base #542: Jump to new versions in Ubuntu

aitormf, 2016-09-07:
# Future jumps of versions in Ubuntu
I first put the task list and then a explanation of how the process will

**Task list:**
- [ ] gazebo7
- [ ] ice-36
- [ ] opencv3
- [ ] ubuntu 16.04

## Gazebo7 
It must be to make small modifications to the plug-ins work with gazebo7, some changes of function names (GetName() to Name(),...), elements that boost, that are now in the library C++ standard and little else.

In his time did a [first version of the plugins](https://github.com/aitormf/JdeRobot/tree/gazebo7.2), but we will have to review them with the updates we have been doing.

##Ice-3.6
We can make the jump without changing any code, the problem is when it comes to Ice 3.6 installation, to install the part of C++ and Java is enough to do the following:

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 5E6DA83306132997

sudo apt-add-repository "deb http://zeroc.com/download/apt/ubuntu$(lsb_release -rs) stable main"

sudo apt-get update

sudo apt-get install zeroc-ice-all-runtime zeroc-ice-all-dev`

But to install Ice for python we need use Pip:

`pip install zeroc-ice`

## Opencv3
The only thing that I can bring on this part is that I've been using it with TeachingRobotics in python on windows and I have not had to touch anything to make it work

## 16.04
To Jump to 16.04, first we need:

- Jump to gazebo7 (Xenial only has gazebo7)
- Compile Openni2 and ardronelib for 16.04 
- Fix a bug that is in [VTK6.2](https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1589847) in the ubuntu repositories. @lr-morales  has done a barckport of VTK6.3 since 16.10 but is incompatible with Dr 16.04 libpcl because it needs vtk6.2

aitormf, 2016-09-13:
With the backports of VTK and PCL made by @lr-morales already not have problems by that side, but now I have seen another new, in Xenial is uses cmake 3.5 and there are changes from the 2.8.

CMake shows me the following (see above all the final warnings):
<pre>
-- The C compiler identification is GNU 5.3.1
-- The CXX compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
version:5.3.1
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gsl'
--   Found gsl, version 2.1
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
/root/JdeRobot/src/libs/colorspaces
/root/JdeRobot/src/libs/jderobotutil
/root/JdeRobot/src/libs/progeo
/root/JdeRobot/src/libs/pioneer
/root/JdeRobot/src/libs/fuzzylib
/root/JdeRobot/src/libs/visionlib
/root/JdeRobot/src/libs/parallelIce
*** ALUT not found
-- Checking for one of the modules 'OPTIONAL;ardronelib'
-- ardronelib NOT FOUND.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
   -- CUDA not found!!
*** CWIID LIBRARIES FOUND
-- Ice
-- Found PY_Ice: /usr/lib/python2.7/dist-packages/Ice.pyc  
-- matplotlib
-- Found PY_matplotlib: /usr/lib/python2.7/dist-packages/matplotlib  
-- numpy
-- Found PY_numpy: /usr/lib/python2.7/dist-packages/numpy  
-- cv2
-- Found PY_cv2: /usr/lib/python2.7/dist-packages/cv2.x86_64-linux-gnu.so  
-- PyQt4
-- Found PY_PyQt4: /usr/lib/python2.7/dist-packages/PyQt4  
-- PyQt4.Qwt5
-- Found PY_PyQt4.Qwt5: /usr/lib/python2.7/dist-packages/PyQt4/Qwt5  
*** Fireware ok
[... 222 lines cut ...]
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/drivers/gazeboserver/plugins/kinect/CMakeLists.txt:18 (add_library):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "kinectPlugin" links to target "Qt5::Widgets" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/drivers/pclRGBDServer/CMakeLists.txt:20 (add_executable):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "pclRGBDServer" links to target "Qt5::Widgets" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /root/JdeRobot/build
</pre>

fqez, 2016-09-14:
I will see you tomorrow, so we can check this issue together! It seems to be failing only with the linking of Qt5::Widgets. If those components work well, we can supress the warnings with ```cmake_policy``` command it seems.

aitormf, 2016-09-19:
Gazebo7 and Ice 3.6 are already tested in 16.04 and run well. But now other dependencies of the project we require to update others that have by have incompatibilities between them.

- VTK5-> VTK7
- PCL-> PCL1.8
- Python2.7-> Python3.5
- Qt4.8-> Qt5.7

Add the new information to the first post

aitormf, 2016-10-04:
PCL1.8, VTK7 and Opencv 3.1 are already tested in 16.04 and run well. 

We have found library to replace qwt, I have added more information in first post

aitormf, 2016-10-05:
I have ported QFlightInstruments (qfi) Altimeter from C++ to python: 
[image]

fqez, 2016-10-24:
Jump done for Ubuntu Xenial 16.04

closed 2016-10-24 by fqez