JdeRobot/base #545: Jump to new versions in Debian

aitormf, 2016-09-12:
# Future jumps of versions in Debian
I first put the task list and then a explanation of how the process will. @lr-morales will take care of this part mainly

**Task list:**
- [ ] gazebo7
- [ ] ice-36
- [ ] opencv3

## Gazebo7 
We need make a gazebo7 package for Debian

## Ice-36
We need make a Ice 3.6 package for Debian

fqez, 2016-09-14:
I think @lr-morales should take the lead with this issue. Moreover I'd like to see how does it do the backports so we can create them for ubuntu if needed.

lr-morales, 2016-09-15:
# Package status for Debian Jessie

## Mandatory packages
*Mandatory* packages to jump to next JdeRobot version.

- [ ] Gazebo (7.3.0)
 - [ ] Solved dependencies
 - [ ] Built x32
 - [ ] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

- [ ] Ice (3.6.2)
 - [ ] Solved dependencies
 - [ ] Built x32
 - [x] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

- [ ] OpenCV (3.1.0)
 - [x] Solved dependencies
 - [ ] Built x32
 - [x] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

## Optional packages
Package building to use versions similar to the JdeRobot Xenial packages

- [ ] Point Cloud Library (1.8.0)
 - [x] Solved dependencies
 - [x] Built x32
 - [x] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

- [ ] The Visualization Toolkit (6.3.0)
 - [x] Solved dependencies
 - [x] Built x32
 - [x] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

## Dependency packages
Packages needed to build the ones in the previous sections.

- [x] CMake (3.6.1)
 - [x] Solved dependencies
 - [x] Built x32
 - [x] Built x64
 - [x] Tested x32
 - [x] Tested x64

- [ ] OpenNI2 (2.2.0.33)
 - [x] Solved dependencies
 - [x] Built x32
 - [x] Built x64
 - [ ] Tested x32
 - [ ] Tested x64

# Extended description

## CMake
* Version used: 3.6.1
* Source from: Debian Strech (Unstable)
* Packages depending
 * The Visualization Toolkit (>= 3.2.0)
 * OpenCV (>= 3.1) 

Built as needed to build other packages. 
Note that build test 7 failed when built, but no issues detected when used in both x32 and x64.

## Gazebo 7
* Version used: 7.3.0
* Source from: Debian Strech (Unstable)

W. I. P.

## Ice
* Version used: 3.6.2
* Source from: *pending* Debian Strech (Unstable) / ZeroC repository 

Solving issue when building packages, most visible in x32.

W. I. P.

## OpenCV
* Version used: 3.1.0
* Source from: OpenCV git repository: https://github.com/opencv/opencv

As we are building from sources, we should make sure that OpenCV is built with similar supports in Jessie and in Ubuntu. Need to confirm which flags are being used in xenial packages (@aitormf ?).

**Note: ** Possible bug when building with both OpenNI and OpenNI2 support. Could you confirm that this "bug" is affecting ubuntu too, @aitormf, @fqez? I'll check it again when building against Jessie x32

## OpenNI 2
* Version used: 2.2.0.33
* Source from: Debian Strech (Unstable)
* Packages depending
 * OpenCV (>= 3.1) 

Built as needed to build other packages. 

## Point Cloud Library 
* Version used: 1.8.0
* Source from: Debian Strech (Unstable)

## The Visualization Toolkit
* Version used: 6.3.0
* Source from: Debian Strech (Unstable)

Had to modify the control file to match Jessie versions whenever possible, so it needs testing to know if works as expected or need to backport related versions.

aitormf, 2017-10-17:
We're not going to keep Debian packages for now

closed 2017-10-17 by aitormf