JdeRobot/base #607: [Deps]: Error installing jderobot-deps from apt-get

almartinflorido, 2016-11-04:
Hi all,

In a clean installation (ubuntu 16.04) I received the next error installing jderobot-deps package from apt-get:

`preparing to unpack .../opencv-dev_3.1.0-1405-g39467d8-dirty_amd64.deb ...
Unpacking opencv-dev (3.1.0-1405-g39467d8-dirty) ...
dpkg: error processing archive /var/cache/apt/archives/opencv-dev_3.1.0-1405-g39467d8-dirty_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/opencv_createsamples', which is also in package libopencv-dev 2.4.9.1+dfsg-1.5ubuntu1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/opencv-dev_3.1.0-1405-g39467d8-dirty_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

fqez, 2016-11-04:
Hi alberto, it does seem that you have a previous installation of OpenCV 2.4 which is incompatible with the new OpenCV 3 used by JdeRobot 5.4.1.
To solve this remove all your OpenCV packages along with jderobot-deps-dev and try to reinstall again.

If you have a previous installation of JdeRobot, you must remove it from your system along with all its dependencies:

      sudo apt-get purge jderobot jderobot-deps-dev
      sudo apt-get autoremove
      sudo apt-get autoclean

Before installing JdeRobot 5.4.1 make sure you have no packages installed from Ice 3.5, OpenCV 2.4, PCL 1.7 and VTK 5.8.

     sudo apt-get purge libopencv* libpcl* libice* libvtk*
     sudo apt-get autoremove
     sudo apt-get autoclean

Then you are good to go.

NOTE: The jderobot-deps-dev package doesn't install ice 3.6 python dependencies (issue #603), so once jderobot-deps-dev is installed follow the instructions from #603 

Sorry for the inconvenience

almartinflorido, 2016-11-04:
Ok you're right. It is a clean installation but I suppose that previously I installed opencv from apt-get.
Now works well.

Thank you

closed 2016-11-04 by almartinflorido

fqez, 2016-11-04:
Remember to install ice 3.6 for python!