JdeRobot/base #608: Simulated Kobuki + KobukiViewer exaple doesn't work

almartinflorido, 2016-11-05:
Hi,

in a clean installation of jderobot I try to run the example Simulated Kobuki + KobukiViewer doesn't work. 
I you excute: 

`gazebo kobuki-simple.world`
and next
`kobukiViewer kobukiViewer.cfg`

You get the next message and the robot doesn't move or show cameras:
`
kobukiViewer kobukiViewer.cfg[Info] loaded Ice.Config file: kobukiViewer.cfg
Pose3D inactive
Camera1 inactive
Camera2 inactive
Laser inactive
Motors inactive
`

If you try with the configuration file exposed in the instructions:
`kobukiViewer.Motors.Proxy=Motors:tcp -h localhost -p 9001
kobukiViewer.Camera1.Proxy=cam_turtlebot_left:tcp -h localhost -p 9001
kobukiViewer.Camera2.Proxy=cam_turtlebot_right:tcp -h localhost -p 9001
kobukiViewer.Pose3D.Proxy=Pose3D:tcp -h localhost -p 9001
kobukiViewer.Laser.Proxy=Laser:tcp -h localhost -p 9001
kobukiViewer.Laser=1
kobukiViewer.Vmax=3
kobukiViewer.Wmax=0.7`

`kobukiViewer --Ice.Config=instructions_cfg.cfg`

You get the next message:
`[Info] loaded Ice.Config file: turtlebot.cfg
Pose3D connected
Outgoing.cpp:478: Ice::ObjectNotExistException:
object does not exist:
identity: cam_turtlebot_left
facet: 
operation: ice_isA
!! 05/11/16 12:09:25.653 error: communicator not destroyed during global destruction.terminate called after throwing an instance of 'IceUtil::ThreadSyscallException'
  what():  ../../include/IceUtil/Mutex.h:303: IceUtil::ThreadSyscallException:
syscall exception: Invalid argument
Aborted (core dumped)`

Finally if you remove the cameras configurations
 from configuration file the example works but yout don't have cameras.

fqez, 2016-11-06:
Hi! yes it's a mistake included with the new turtlebot version. Just change cam_turtlebot_left and cam_trutlebot_right by CameraL and CameraR respectively in kobukiViewer.cfg file.

almartinflorido, 2016-11-06:
Ok, Thank you @fqez

closed 2016-11-06 by almartinflorido