JdeRobot/base #618: ColorTuner with python

aitormf, 2016-11-09:
We need make Color Tuner tool with python for use it with TeachingRobotics in others plataforms like Windows

aitormf, 2016-11-14:
I have already finished the HSV filter but the slider values are different from the colorTuner of c ++.
I have followed [this example](http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_imgproc/py_colorspaces/py_colorspaces.html) to make it.

- In python H goes from 0 to 179 but in c++ goes to 6.28 (shown in the sliders)
- In python S goes from 0 to 255 but in c++ goes to 1 (shown in the sliders)
- V is the same in the two

Is any problem for this?

jmplaza, 2016-11-15:
The ranges in C++ are right according to the intuition: H means angle, and should range from 0 to 2*PI (=6.28) to cover the whole circle, S means radius and a good convention is to range it from 0 to 1. 
Other good values for their ranges in Python are those to be used in the opencv function to apply the color filter (cv.inRange) over HSV images. I'm not sure about these ranges, maybe 0..255?

jmplaza, 2016-11-15:
For the moment I would put exactly the same values than in the C++ version of this tool. Maybe we should think in keeping only one of them, the Python one, in the long term.

jmplaza, 2016-11-15:
Uhm, not quite sure about the last comment. According the opencv-3.0.0 doc you pointed out @aitormf : "For HSV, Hue range is [0,179], Saturation range is [0,255] and Value range is [0,255]. Different softwares use different scales. So if you are comparing OpenCV values with them, you need to normalize these ranges."

Maybe a better choice is to use in the GUI the same values cv.inRange is going to use for HSV images... More opinions @almartinflorido @eperdices ?

closed 2016-11-17 by fqez