JdeRobot/base #623: Include cmake config files

chanfr, 2016-11-11:
In order to use jderobot from external projects the idea is to include both JdeRobotConfig.cmake and JdeRobotConfigVersion.cmake. Including those files, external projects will be able to use jderobot with built-in cmake support simply using:
FIND_PACKAGE(JdeRobot REQUIRED)

chanfr, 2016-11-11:
@fqez @jmplaza please take a look on this issue and please verify if it works for you too. I cannot verify the package because I am using several custom version of the dependencies.

fqez, 2016-11-11:
Hi @chanfr I haven't tested it yet (doing it this weekend), but checking the PR #626 I see that you have removed easyiceconfig_cpp and cvblob libraries. Is that correct? :S

chanfr, 2016-11-12:
Hi @fqez, I didn't remove those libraries y simply renamed them. In order to simply the generation of the cmake config file I needed to rename the folder of the library with the same name of the library, so the target of easyiceconfig_cpp is easyiceconfig, the target of cvBlob is cvblob, and so...

chanfr, 2016-11-12:
sorry, you are right, the PR is wrong :-( I will create it again... I don't know why the paths with the new name were not included.

chanfr, 2016-11-12:
fixed

chanfr, 2016-11-12:
To test if this if working, simply compile the code and install it, then create a cmake with:

` cmake_minimum_required(VERSION 2.8)

include(FindPkgConfig)

FIND_PACKAGE(JdeRobot REQUIRED)

message("JdeRobot_INCLUDE_DIRS: ${JdeRobot_INCLUDE_DIRS}")
message("JdeRobot_LIBRARIES: ${JdeRobot_LIBRARIES}")`

and execute  cmake .

fqez, 2016-11-21:
Hi @chanfr I've been testing it and it compiles well, but when I try to install it (I've compiled the project out-of-source) it fails... 

This is the trace:

```
CMake Error at cmake_install.cmake:2536 (include):
  include could not find load file:

    /home/fqez/github/testing/jderobot/JdeRobotInstallConfig.cmake

Makefile:115: fallo en las instrucciones para el objetivo 'install'
make: *** [install] Error 1
```

If i manually copy JderobotInstallConfig.cmake to the root directory it fails again with the following trace:

```
CMake Error: File /home/fqez/github/testing/jderobot/build/cmake/JdeRobotConfig.cmake.in does not exist.
CMake Error at /home/fqez/github/testing/jderobot/JdeRobotInstallConfig.cmake:17 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  cmake_install.cmake:2536 (include)

Makefile:115: fallo en las instrucciones para el objetivo 'install'
make: *** [install] Error 1
```

I think that out-of-source build is not supported as long as ~/Jderobot/cmake directory is not copied into ~/Jderobot/build, which the process is complaining for.

Copying the cmake directory into build manually solves the problems and the installation finish as expected. The CMakeLists.txt test file works fine also.
Could you re-check this please?

Good job anyway :D

chanfr, 2017-01-22:
You are right installation from a custom build path was not working. Fixed!

chanfr, 2017-01-24:
@fqez can you check if everything is working now?

fqez, 2017-01-28:
Confirmed, it works now!

closed 2017-01-28 by fqez