JdeRobot/base #641: [Packages] We need to make some custom ROS packages

fqez, 2016-11-24:
The current ROS packages provided by the OSRF are compiled and work with python2.7. JdeRobot does not support python2.7 any longer in benefit of python3.5. This version jump breaks the previous ROS Compatibility Library, so we need to compile the Library needed ros packages with python3 to make it works.

fqez, 2016-11-29:
The current state of the packages is "testing". We do have some custom-compiled ROS packages with python3 so we are now testing the installation of those packages. To be more specific:

ROS debian custom packages created:
 
* [x] xmlrpcpp
* [x] catkin
* [x] class-loader
* [x] cmake-modules
* [x] cpp-common
* [x] gencpp
* [x] geneus
* [x] genlisp
* [x] genmsg
* [x] gennodejs
* [x] genpy
* [x] geometry-msgs
* [x] message-filters
* [x] message-generation
* [x] message-runtime
* [x] pluginlib
* [x] rosbuild
* [x] rosclean
* [x] rosconsole
* [x] roscpp
* [x] roscpp-serialization
* [x] roscpp-traits
* [x] rosgraph
* [x] rosgraph-msgs
* [x] roslang
* [x] roslaunch
* [x] roslib
* [x] rosmaster
* [x] rosout
* [x] rospack
* [x] rosparam
* [x] rospy
* [x] rostest
* [x] rostime
* [x] rosunit
* [x] sensor-msgs
* [x] std-msgs
* [x] xmlrpcpp

# First Approach

## Package generation

The current custom packages does not work as expected. The "main" package (catkin) will not install all the setup files required for sourcing ROS packages. We are now trying to build them correctly, as we test our bilingual tools. The first test, with the ROS packages installed in a local environment works as expected, [here](https://www.youtube.com/watch?v=kjTnhAaLmr4) is a video showing the example running.

@shady-cs15 As soon as the packages are correct and released, we will let you know.

shady-cs15, 2016-12-04:
Good to see :) @fqez

closed 2017-02-23 by fqez