JdeRobot/base #644: [drivers] support for new RPLIDAR 360º laser scanner

jmplaza, 2016-12-03:
This time we are not going to develop our own driver over the manufacturer software [RoboPeak](https://github.com/robopeak/rplidar_ros) we are going to use ROS driver and our (on going) compatibility library for ROS-messages. This low cost [sensor](http://www.robopeak.com/blog/?cat=5) delivers 2D scan with distances to obstacles (<6m) around (360º).

We will have to extend our ROS compatibility library to support ROS messages from this sensor.

aitormf, 2016-12-12:
I have followed [this](http://wiki.ros.org/rplidar) instructions to try the sensor with official ROS Kinectics and it works well.

Also, I have created debian necessary packages of RPLIDAR for own ROS Kinectics version

aitormf, 2016-12-16:
We need update Ice laser interface to support all necessary information of [ROS LaserScan message](http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html). I have thought in:

<pre>
Laser {
    int[] data
    int numLaser
    float min_angle
    float max_angle
    float max_range
    float min_range
   time_stampt
}
</pre>

aitormf, 2016-12-29:
I have a first approximation: rplidar_ros_node + kobukiViewer vs rplidar_ros_node + rviz:

[image]
I need put own ros laser listener in a separate thread to not block the entire Tool. Also we should think on paint points instead of lines.

closed 2017-02-23 by fqez