JdeRobot/base #659: Camera of a Kobuki robot in Gazebo (with ROS plugins) connected to cameraview tool

jmplaza, 2016-12-28:
Related to #639 , that issue has been already solved with a real USB camera and a ROS node, as can be seen at https://www.youtube.com/watch?v=kjTnhAaLmr4  It uses regular usbcamera topic, not any adhoc ROS messages (as we had before).

The next step in convergence towards ROS is to use that (bilingual) cameraview component to display the images from the camara of a Kobuki robot in Gazebo. A Gazebo with ROS plugins (not our JdeRobot ones).

Some custom ROS debian packages (with Kinetic) have been created to simplify the installation of the ROS stuff for our use in JdeRobot (#641).

shady-cs15, 2017-01-08:
@fqez Just to be sure, the ROS compatibility library integration is in the 'fqez-cameraviewROS' branch right? Because in that branch the 'roscompat' folder isn't present in the libs directory.
Am I missing something?

fqez, 2017-01-12:
@shady-cs15 yes! the state of the art of the library is in the branch fqez-cameraviewROS. I will merge it ASAP to make things easier for everyone. Hope you are having no problems with it.

commit 73cd0064 (2017-03-13): Merge pull request #720 from fqez/issue_639

[issue #639] Made cameraview bilingual
Fix #659 #639
closed 2017-03-13 by fqez