JdeRobot/base #660: JdeRobot logo in the GUI of all the tools of the project

jmplaza, 2016-12-29:
Let's put the logo of our project (the four colored circles) in the GUI of all the JdeRobot tools. Let's start with the most used ones:
- [ ] cameraview
- [ ] colortuner
- [ ] opencvdemo
- [ ] kobukiViewer
- [ ] uav_viewer
- [ ] rgbdviewer
- [ ] visualHFSM

shivangg, 2017-04-15:
`uav_veiwer/gui/mainwindow.ui` seems to be depreciated because running `uic` generates `ui_mainwindow.h` that is different from the `uav_viewer/gui/ui_mainwindow.h` . `velBox` and `dataBox` are misssing. `mainwindow.ui`should be updated to the suitable version (to the one that generated the current `ui_mainwindow.h` ) to finally add the Jderobot icon.

jmplaza, 2017-04-17:
Hi @shivangg, 

uav_viewer has been recently improved in the official repository solving issue  #702. Are you running the new release of uav_viewer? Please update your git clone an run the new one.

You are right, the JdeRobot icon seems to be missing :-)

shivangg, 2017-04-19:
Hi @jmplaza,
I was referring to the C++ version of uav_viewer but as mentioned by you in issue #775 that it is depreciated and the python version will be the preferred release. In this case, what will be the fate of the C++ version? Should the JdeRobot logo be added to it?

jmplaza, 2017-04-19:
As the C++ component is now deprecated, maybe adding the logo is not worth the effort

okanasik, 2017-05-25:
On visualHFSM, where would be more appropriate to put the logo? I also checked the "about" menu, but it is not implemented. Maybe we can create a dialog window and put the logo here. What do you think?

jmplaza, 2017-05-25:
Take a look at where the logo is locate in other tools like [ColorTuner] (http://jderobot.org/index.php/Tools#ColorTuner_.28Python.29). Maybe on the front panel, near the corner is an elegant place. Feel free to enlarge the GUI window a little bit or reduce the logo size at will.

closed 2021-02-21 by jmplaza